Pmatsol58/VyperCE6.1 #3: Strategy for getting rid of STM licensing issues

seho85, 2022-02-14:
Hiho,

i'd had a look into the src, and tried to figure out where the actual licensing issues were located. And how much effort it would be to get rid of the licensing issues. But i'm pretty sure some of the other developers has done something similar.

I saw, that the calls to the STM32 HAL were basically located in the _Marlin/src/HAL/STM32_ and _STM32F1_ Folders. Also I saw somewhere a few calls to the Arduino Library in form of analogRead(). 

But currently I haven't a good overview about Marlin and got absolutely no idea which peripherals of the MCU were used. I'm pretty new to 3D Printing, but I have done a few hobby projects in the STM32 ecosystem and I'm willed to put work into getting rid of the licensing issues. But for now I haven't done anything with PlatformIO yet and I need to dig into that system.

Re implementing the whole STM32 will definitely be a bigger task. Fortunately there is already a platform for Platform IO which seams to support the GD32F103RE. And there is also an Arduino package that has also support for the GD32F103RE. 

https://github.com/orgs/CommunityGD32Cores/repositories

Maybe this could be a starting point for getting rid of the licensing issues.

Currently I have no overview which of the peripherals of the actual MCU were used in the Marlin Firmware. 

But my Idea is first create a Listing of all used Peripherals which needs to be reintegrated and create a new GD32 HAL component in Marlin that just supports all the peripherals used by the MCU on the Anycubic Vyper, or to be more exact the TriGorilla V0.0.6 Printer mainboard. 

Please see this just as some brainstorming what we would be able to do. I don't want to tell anybody what he needs to do.

Any thoughts about that were welcome. Also when you think that's an outlandish idea.

Kind Regards,
Sebastian

Pmatsol58, 2022-02-14:
Any ideas are welcome.
I have been looking through the framework at the licenses and most of them, especially in the HAL, are open with either 3-clause BSD or Apache licenses. The areas of concern are CMSIS and USB which still reference the restrictive license.

Looking at later version frameworks the CMSIS has also moved to open licenses so the USB is the only area of concern, it looks like they are gradually moving to an open framework. 

Vyper does not use USB and I am not sure it uses any CMSIS either since we do not build the boot loader, so I am inclined to think that it is not an issue for Vyper, but it may be an issue for the main stream Marlin since USB can be enabled in other configurations. Maybe changes in the sanity checking could be made to at least warn users of the possible license issue.

Best regards
PaulM.

seho85, 2022-02-15:
Thanks for your response.

Yep, your'e right. I got something wrong with the HAL License. I just downloaded the _SL0044_ and thought it's also related to HAL.

Here is the overview from STM about there licenses in the STM32 F1 HAL:

| Component                       | License              | Copyright |
|:---------                       |:-------              |:----------|
| CMSIS                           | Apache License 2.0   | Copyright (c) 2009-2017 ARM Limited. All rights reserved. |
| CMSIS Device                    | Apache License 2.0   | ARM Limited - STMicroelectronics |
| STM32F1 HAL                     | BSD-3-Clause         | STMicroelectronics |
| BSP STM32F1xx_Nucleo            | BSD-3-Clause         | STMicroelectronics |
| BSP STM32VL-Discovery           | BSD-3-Clause         | STMicroelectronics |
| BSP STM3210C_EVAL               | BSD-3-Clause         | STMicroelectronics |
| BSP STM3210E_EVAL               | BSD-3-Clause         | STMicroelectronics |
| BSP Components                  | BSD-3-Clause         | STMicroelectronics |
| BSP Adafruit                    | BSD-3-Clause         | STMicroelectronics |
| FreeRTOS kernel                 | MIT                  | Copyright (C) 2017 Amazon.com, Inc. or its affiliates |
| FatFS                           | BSD-3-Clause         | ChaN - STMicroelectronics |
| LwIP                            | BSD-3-Clause         | Swedish Institute of Computer Science  |
| STM32_USB_Device_Library        | ST SLA0044           | STMicroelectronics |
| STM32_USB_Host_Library          | ST SLA0044           | STMicroelectronics |
| STemWin                         | ST SLA0044           | STMicroelectronics |
| STM32 Projects                  | ST SLA0044 (BSD-3-Clause for basic Examples) | STMicroelectronics |
| STM32 Utilities                 | BSD-3-Clause         | STMicroelectronics |

from https://github.com/STMicroelectronics/STM32CubeF1/blob/master/License.md

So basically everything licensed with _ST SLA0044_ is be a problem.

From _SLA0044_:

<pre>
4. This software or any part thereof, including modifications and/or derivative works of this software, must be used and
execute solely and exclusively on or in combination with a microcontroller or microprocessor device manufactured by or for
STMicroelectronics.</pre>

So from this point it still looks better then I thought.

I also think the mainboard doesn't use the USB capabilities, the USB connector on printer self is realized due a CH431 USB-TTL. I just rechecked this on the photos i took from my mainboard.

If we really would need to implement the USB-Device features, there a few example from GD32 firmware librrary arround the USB-Device stack. https://www.gigadevice.com/firmware/gd32f101xxxx-firmware-library/

But this sounds at all like a much smaller task then I assumed it would be.

I'll have a look and try to figure out where the USB-Device stuff is used in Marlin. And I'll also see how to realize this warning about breaking license conditions.

And maybe we can start also a talk to the maintainers of the marlin project, to get an idea what they think about that idea. 

Have you talked to them at some point before? I just heard from Sebastian Dammann that the maintainers just closed any issues about integrating the GD32.

Pmatsol58, 2022-02-15:
You are correct about the USB being realized by a CH341 but there are resistor options on the board to bring the USB signals directly to the connector although I do not see them changing the build to do this any time soon. 

Yes the more I look into it too the more I think it is less of an issue and am tempted to issue a binary update. As I mentioned, the framework being used at present is older and does still reference the SLA0044 for the CMSIS too but I don't think we are using any of that in the build, it would be nice to use the later framework but that fails to compile since it was changed to support variants differently, I'm sure it can be used but I do not understand the Arduino changes that it supports and so haven't given it the time it needs yet.

I haven't started any conversations with Marlin maintainers yet although I have seen the discussions that Seb had with them.

Ideally at the end of the day I would like to use the latest Marlin build with the latest framework since that should give a better release with all the fixes I am sure exist in the present code (there are over 800 files different between 2.0.9.1 and 2.0.9.2 and we have a 2.0.9.3 now), it just seems a monumental task at present.

seho85, 2022-02-16:
I digged a little bit more trough the codebase but there a few things that weren't completely clear for me. I never worked with PlatformIO before and need to figure out more exactly where the different components where coming from and how they work together.

> As I mentioned, the framework being used at present is older and does still reference the SLA0044 for the CMSIS too ...

The framework your are talking about is part of the Platform.IO stuff that is downloaded when you were building the firmware? Or is it included within the Marlin Project?

I had also a little deeper look into the GD32 stuff that is available for Plattform IO. Unfortunately the Arduino stuff is still under development and nothing has been realized for GD32F1xx derivates. But it could be just a matter of time till there is coming some more. From what i've seen GD came up with the release of there firmware on Jun 2021.

Currently I'm trying to create "new" plattform, basically a clone of the STM32 platform in Marlin, that can be used at a starting point to move codebase to the GD32 plattform of PlatformIO. 

Still an open question for me is if the support for the Arduino Platform is a must. Or if it is possible to map everything to functions from the GD32 platform of PlatformIO.

seho85, 2022-02-17:
Thanks for the explanation.

I'm currently using like in your explanation the
> platform         = ststm32@~12.1

I also searched trough the platform directories, and the only files where SLA0044 license is mentioned is by the following components:

ststm32 platform (v12.1.1):
> platforms/ststm32/examples/stm32cube-hal-iap/
> platforms/ststm32/examples/stm32cube-hal-usb-device-dfu/

So in the 12.1.1 Platform the SLA0077 is only used for the examples. Not for the actual libraries if understand it correctly.

ststm32 platform (v10.0.1):
> platforms/ststm32@10.0.1/examples/stm32cube-hal-usb-*
> platforms/ststm32@10.0.1/examples/stm32cube-hal-iap/

So in the 10.0.1 Platform the SLA0077 is only used for the examples. Not for the actual libraries if understand it correctly. Same as for 12.1.1

I searched in the marlin source and iap (in application programming) isn't used. But I'm not sure about this, because somehow the new firmware binary must been brought from the SD to the internal flash of the MCU. But I think, that is some we don't need to care about, because the binary created by this project doesn't do anything like that.

framework-arduinoststm32 (v?.?.?)

> packages/framework-arduinoststm32/cores/arduino/stm32/usb/*
> packages/framework-arduinoststm32/system/Middlewares/ST/STM32_USB_Device_Library/*
> packages/framework-arduinoststm32/system/Middlewares/ST/STM32_USB_Host_Library*

So it's basically like you mentioned in the posts before: the USB libraries. But I haven't found anything CMSIS related.

I think, if we manage to get the whole USB stuff excluded from the binary, then we should have get rid of the licensing issues. But it would be good if someone else could recheck this.

seho85, 2022-02-17:
Using the latest changes from "integrate_gd32" branch, and the changes attached in the tweakds_to_arduino_fw.txt, the compiler doesn't show anything compiled that belongs to usb anymore.

```
Processing GD32F103 (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/GD32F103/FrameworkArduinoVariant/PeripheralPins.c.o
Compiling .pio/build/GD32F103/FrameworkArduinoVariant/variant.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/HardwareTimer.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/Print.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/RingBuffer.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/USBSerial.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/VirtIOSerial.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/WInterrupts.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/WSerial.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/WString.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/abi.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/avr/dtostrf.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/board.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/hooks.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/itoa.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/main.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/new.cpp.o
Compiling .pio/build/GD32F103/FrameworkArduino/pins_arduino.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/device.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/generic/condition.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/generic/cortexm/sys.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/generic/generic_device.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/generic/generic_init.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/generic/generic_io.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/generic/generic_shmem.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/generic/time.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/init.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/io.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/log.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/libmetal/shmem.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/mbox_ipcc.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/open-amp/remoteproc/remoteproc_virtio.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/open-amp/rpmsg/rpmsg.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/open-amp/rpmsg/rpmsg_virtio.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/openamp.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/rsc_table.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/virt_uart.c.o
Compiling .pio/build/GD32F103/FrameworkArduino/stm32/OpenAMP/virtio/virtio.c.o
[... 367 lines cut ...]
Indexing .pio/build/GD32F103/libba3/libIWatchdog.a
Archiving .pio/build/GD32F103/liba79/libSPI.a
Indexing .pio/build/GD32F103/liba79/libSPI.a
Archiving .pio/build/GD32F103/lib331/libWire.a
Indexing .pio/build/GD32F103/lib331/libWire.a
Archiving .pio/build/GD32F103/lib862/libSoftwareSerial.a
Indexing .pio/build/GD32F103/lib862/libSoftwareSerial.a
Linking .pio/build/GD32F103/main_board_20220217-133753.elf
Checking size .pio/build/GD32F103/main_board_20220217-133753.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  29.0% (used 19024 bytes from 65536 bytes)
Flash: [====      ]  35.3% (used 185188 bytes from 524288 bytes)
Building .pio/build/GD32F103/main_board_20220217-133753.bin
================================================= [SUCCESS] Took 11.51 seconds =================================================

Environment    Status    Duration
-------------  --------  ------------
GD32F103       SUCCESS   00:00:11.513
================================================== 1 succeeded in 00:00:11.513 ==================================================

```

Unfortunately that requires to tweak the arduino sources, but maybe we can find a way to get around this.

[tweaks_to_arduino_fw.txt](https://github.com/Pmatsol58/VyperCE6.1/files/8088884/tweaks_to_arduino_fw.txt)

seho85, 2022-02-17:
I had also a nice talk to @Sebazzz which brought me to the problems that Marlin Devs have.

And one of their statements were:
> no the peripheral driver us just the code that runs on the mcu that allows the usb to work its part of stms arduino framework, they would have to strip all usb support from the stm framework

and

> if any of the usb features are built into the binary, whether used or not
if the usb driver (and anything else under the more restrictive licensing that I don't know about) was stripped from stms arduino framework and not built into the binary it would be fine

Thats basically what the tweak_to_arduino_fw.txt patch would do. Maybe I overlooked some files related to USB. But from what I understand, is that when you were using a patched platform (that ensures nothing USB related is in it) for your build you can continue your work. And distribute the binary package.

Maybe I'll continue working on switching more stuff to GD32 Platform in the future, but for now I have a few other tasks to solve.

Sebazzz, 2022-02-17:
Possibly relevant: STMicroelectronics/STM32CubeH7/issues/139

Sebazzz, 2022-02-17:
At this point it is becoming a bit of a minefield though.

Pmatsol58, 2022-02-18:
It is a minefield. If we could reduce the framework to just the parts that are needed such as the HAL then it should be ok since that is under a BSD license but I don't know enough about platformIO and the scripts it uses or the Arduino requirements for it to build properly.

seho85, 2022-02-18:
It definitely is. But at least the dawn has come that we can see a little bit more.

@Pmatsol58 
I just deleted files from the Platform Directory, and the build looked trough the outputs on build in VSCode if they were not build anymore. 

And I also excluded the usb stuff from the copied stm32 folder in Marlin/src/HAL, in the gd32f1.ini that I've created.

Its also thinkable to reverse the process, delete everything and the re-add the files after a review about license to the platform directory. Build and see why it fails. Repeat that until build finally succeeds.

seho85, 2022-02-19:
The whole ethernet stuff is not enabled due _HAL_ETH_MODULE_ENABLED_ is not defined.

And I couldn't find lwIP either in any of the platform folders of the stm32. Only in the by default installed for the avr.

I pushed the whole tweaked framework-arduinoststm32 folder into a git repository that available on my github page. https://github.com/seho85/framework-arduinoststm32

This folder must overwrite the <Home>.platformio/packages/framework-arduinoststm32 folder.

Pmatsol58, 2024-09-04:
Development on this version has ceased, see https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2

closed 2024-09-04 by Pmatsol58