Pmatsol58/VyperCE6.1 #6: Printing from host (Octoprint) doesn't show progress

Dominik-1980, 2022-02-15:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

When printing from Octoprint no progress is shown in the display. 
It stays at 0%

### Version of CR6Community Firmware

6.1c

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Other (I will explain below)

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

Pmatsol58, 2022-02-16:
If I remember this requires a setting in the slicer or an octoprint plugin, check out this blog post from Sebastiaan: https://damsteen.nl/blog/2021/03/13/estimated-time-remaining-with-the-cr6-community-firmware

yasars, 2022-02-16:
can we fix "printing from host" on the display??

Pmatsol58, 2022-02-18:
I don't use octoprint so I'm not sure if name can be used in place of "printing from host", there probably is a way of doing it but I just don't know.

closed 2022-02-18 by Dominik-1980