Pmatsol58/VyperCE6.1 #7: Z-offset is not working

spacejay01, 2022-02-16:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

When I change the Z-offset during the print via Tune, the Z-offset is working. The value is also stored and visible via the calibration menue. But when I start another print the offset is ignored. Even if I set a offset via the calibration menue.

### Bug Timeline

did not see this behavior with the stock firmware

### Expected behavior

Excpectation is that the Z-offset is used for each print job

### Actual behavior

Z-offset is just ignored 

### Steps to Reproduce

1. Start printing
2. change offset via Tune
3. the printer changes the offset
4. the printjob is finished
5. start a new print job
6. the offset is ignored

### Version of CR6Community Firmware

6.1c

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Other (I will explain below)

### Your Slicer

Cura

### Host Software

SD Card (headless)

Pmatsol58, 2022-02-16:
This is a known issue. Work around is to level the printer with a 0.0 Z offset, start printing and adjust Z offset as required, then perform another level which will incorporate the offset into the mesh. Any prints following this will use the correct Z offset.

rommulaner, 2022-03-16:
I also mentioned this issue the last days while switching to this CE Firmware. Is it possible to make a workaround by modifying the start gcode by setting the z-offset value by using the actual LCD value (like Z_OFFSET or whatever the variable called) and set it after the homing command?
I don't know the correct variable name.

vito30x, 2022-03-20:
> I also saw this issue the last days while switching to this CE Firmware. Is it possible to make a workaround by modifying the start gcode by setting the z-offset value by using the visible LCD Z-Offset value (like M428 Z{Z_OFFSET} or whatever the variable called) and set it after the homing command? I don't know the correct variable name.
> 
> Edit: Got it done by myself. Now issue is solved. Can be downloaded on https://GitHub.com/rommulaner
Hi, Danke für den Fix. Habs bei mir implementiert.
Du hast auch die EstepsHandler.cpp geändert.
`sprintf_P(cmd, PSTR("G1 E%s **F90**"), str_temp);` Welchen Grund hat es den F-Wert von 50 auf 90 zu erhöhen?

rommulaner, 2022-03-20:
Stimmt, hatte ich etwas erhöht, da mir das zu langsam ablief. Passt so besser zur realen Druckgeschwindigkeit.

Pmatsol58, 2024-09-04:
Development on this version has ceased, see https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2

closed 2024-09-04 by Pmatsol58