Pmatsol58/VyperCE6.1 #8: Set Z=0 to the middle of the Bed before leveling...

Dominik-1980, 2022-02-18:
### Describe the feature you want

If I do a bed leveling, it shows me values between -2,4 and -2,00.
It would look much better in the leveling mesh if I could set to Z=-2,2.
This is approximately the center of the bed.

The firmware could take one probe at the center of the bed and that Z=0 to this value,
then do the automatic bed leveling...

Pmatsol58, 2022-03-03:
There is a build option to use the nozzle as the Z endstop for homing and this will probe the centre of the bed when homing the Z axis, leveling will then be based around this figure giving values closer to 0. There are disadvantages to this too in that you cannot home with anything on the bed and a G34 has to be used to level the two Z screws rather than use the Z optical stops on each side of the bed.

Dominik-1980, 2022-03-03:
Okay. Thanks for your response.

I will leave it as it is.
The disadvantages of homing with the probe prevail,
just to look nicer on the leveling...

closed 2022-03-22 by Dominik-1980