Pmatsol58/VyperCE6.1 #11: M300 frequency and tone queue not working

Dominik-1980, 2022-02-19:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

M300 always plays the same tone. I can change duration with P, but frequency with S doesn't work.
Also the tone queue doesn't work. 
I have a list of M300 commands at the End-gCode, to play a melody, but it just plays one tone.
Always the same beep with the same frequency as the touchscreen input uses...

### Version of CR6Community Firmware

Anycubic Vyper CE6.1c

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Other (I will explain below)

Pmatsol58, 2022-03-03:
The M300 beep uses the tone output of the display which is a fixed frequency, I need to look at bringing it back to the onboard speaker which can have frequency set.

smcbride-ca, 2022-04-20:
[This commit](https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.1d/commit/b49f8c4315c130473a49057260fa1f6880cae1fa) fixes the issue. If these changes are made we can get sound working properly

Pmatsol58, 2022-05-22:
Committed.

closed 2022-05-22 by Pmatsol58