Pmatsol58/VyperCE6.1 #14: Prints are too tall - Don't match slicer

smcbride-ca, 2022-03-20:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

The height of objects that have been printed is taller than shown in the slicer. This only began after switching to CE firmware. Example: Calibration cube, 20 * 20 * 20. Stock firmware produces a cube ~20 * ~20 * 20. CE Firmware produces a cube ~20 * ~20 *21.5mm. This occurs on all models tested

All stock hardware
[Calibration cube.gcode.zip](https://github.com/Pmatsol58/VyperCE6.1/files/8311739/Calibration.cube.gcode.zip)

### Bug Timeline

After switching from stock to CE

### Expected behavior

The height of the object should match the slicer.

### Actual behavior

The height of the object is taller than in the slicer

### Steps to Reproduce

1. Use CE firmware 6.1c
2. Print an object

### Version of CR6Community Firmware

6.1c

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Other (I will explain below)

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

smcbride-ca, 2022-03-20:
No change while using https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.1d

rommulaner, 2022-03-21:
I've just uploaded my reworked 6.1d. The 1.5mm issue should now be solved with that. I did't have time to print a cube today.

smcbride-ca, 2022-03-21:
Fixed in the new 6.1d revision. Closing the issue.

closed 2022-03-21 by smcbride-ca