Pmatsol58/VyperCE6.1 #16: [FR] Anycubic Kobra

ckarcz, 2022-05-02:
### Are you looking for hardware support?

Any ideas on getting this to work on the Kobra Max (which is basically a large Vyper)? 
Would simply updating the Configuration.h work?

### Describe the feature you want

Anycubic Kobra MAX support.

rommulaner, 2022-05-22:
I don't know the exact data of the Kobra Max, but if it is the same setup and parts it is easily possible, yes.

Pmatsol58, 2022-05-22:
I believe the motherboard is different and so it may not work. Anycubic have not released hardware details or source as far as I know and I do not have access to a Max and so cannot support it.

ckarcz, 2022-05-22:
the motherboard for the Vyper is a TriGorilla+ and the motherboard for the
Kobra appears to be a Trigorilla Pro A.

rommulaner, 2022-05-22:
Has the Kobra Max the same board as the Kobra?

Pmatsol58, 2024-09-04:
It is possible but without a Kobra to work with it will not happen.
Development on this version has ceased, see https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2

closed 2024-09-04 by Pmatsol58