Pmatsol58/VyperCE6.1 #19: Linear Advance K value not saved

Pmatsol58, 2022-09-08:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

When using the screen to adjust the K value for linear advance whilst printing it is not saved in the eeprom. Using M900 Kx followed by M500 works but the screen value is not saved.

### Bug Timeline

New

### Expected behavior

It should be saved to the eeprom as the z offset value is when it is changed via the screen.

### Actual behavior

Screen shows new value but it is not saved to eeprom, next time the printer is powered on it reverts to the previous value that was saved in eeprom, usually 0.0

### Version of CR6Community Firmware

6.1d

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Other (I will explain below)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

brusyev, 2023-09-03:
I am seeing the same issue on version 6.1e but with bed mesh leveling...values get wiped after power cycle.

rommulaner, 2024-01-18:
solved with 6.2.

brusyev, 2024-01-18:
> solved with 6.2.

Thank you so much for continuing to improve this firmware.

rommulaner, 2024-01-18:
Thank You! I appreciate that.
I will try to make some changes on the nozzle offset and bed mesh values to get an setup that works for every Vyper. Currently the different end stop positions at Z and the different nozzle positions in the heat block can create high bed mesh values of 1 or 2 or more mm. With this values it is not possible to use FADE HEIGHT without getting bad layers and big deviations in height of the part. That's not how bed leveling is planned to work. I'll try to implement a first manual adjustment of the nozzle height that will be saved permanently and after that you can start with the bed leveling. Then it works as it should. This offset can also cause the bed leveling to stop because the nozzle is out of range.
Let's see whats possible...

If you want to support me, you can donate over paypal.me/robertrwwagner 

Thank You! :)

Pmatsol58, 2024-09-04:
Development on this version has ceased, see https://github.com/rommulaner/Anycubic_Vyper_MB_CE_6.2